ansys/pyedb-core #368: Large message issue.

vmartidis, 2024-02-08:
### 🔍 Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### 🐞 Description of the bug

In case of large designs Layout.primitives property fails.
primitives property calls _get_items which tries to retrieve a large message. 
[image]

### 📝 Steps to reproduce

Call layout.primitives in a large design.

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

24R2

### 🐍 Which Python version are you using?

3.8

### 📦 Installed packages

```shell
alabaster==0.7.12
ansys-api-edb==1.0.0                                                                    
ansys-edb-core
ansys-tools-protos-generator==0.1.0                                 
anyio==4.2.0                                                        
argon2-cffi==23.1.0                                                 
argon2-cffi-bindings==21.2.0                                        
arrow==1.3.0                                                        
asttokens==2.4.1                                                    
async-lru==2.0.4                                                    
atomicwrites==1.4.0
attrs==23.2.0
Babel==2.10.1
backcall==0.2.0
beautifulsoup4==4.11.1
bleach==4.1.0
build==0.7.0
cachetools==5.0.0
certifi==2021.10.8
cffi==1.16.0
charset-normalizer==2.0.12
colorama==0.4.4
comm==0.2.1
contourpy==1.1.1
coverage==6.3.2
cycler==0.12.1
debugpy==1.8.0
decorator==5.1.1
defusedxml==0.7.1
distlib==0.3.4
docutils==0.17.1
et-xmlfile==1.1.0
exceptiongroup==1.2.0
executing==2.0.1
fastjsonschema==2.19.1
filelock==3.6.0
flit==3.7.1
flit_core==3.7.1
fonttools==4.48.1
fqdn==1.5.1
google-api-core==2.7.1
google-api-python-client==2.40.0
google-auth==2.6.0
google-auth-httplib2==0.1.0
googleapis-common-protos==1.55.0
grpcio==1.47.0
grpcio-tools==1.47.0
h11==0.14.0
httpcore==1.0.2
httplib2==0.20.4
httpx==0.26.0
idna==3.3
imagesize==1.3.0
importlib-metadata==4.11.3
importlib-resources==6.1.1
iniconfig==1.1.1
ipykernel==6.29.1
ipynbname==2023.2.0.0
ipython==8.12.3
isoduration==20.11.0
jedi==0.19.1
Jinja2==3.1.2
json5==0.9.14
jsonpointer==2.4
jsonschema==4.21.1
jsonschema-specifications==2023.12.1
jupyter-events==0.9.0
jupyter-lsp==2.2.2
jupyter_client==8.6.0
jupyter_core==5.7.1
jupyter_server==2.12.5
jupyter_server_terminals==0.5.2
jupyterlab==4.1.0
jupyterlab_pygments==0.3.0
jupyterlab_server==2.25.2
keyring==23.5.0
kiwisolver==1.4.5
MarkupSafe==2.1.1
matplotlib==3.7.4
matplotlib-inline==0.1.6
mistune==3.0.2
nbclient==0.9.0
nbconvert==7.15.0
nbformat==5.9.2
nest-asyncio==1.6.0
notebook==7.0.7
notebook_shim==0.2.3
numpy==1.24.4
numpydoc==1.3.1
openpyxl==3.1.2
overrides==7.7.0
packaging==21.3
pandas==2.0.3
pandocfilters==1.5.1
parso==0.8.3
pep517==0.12.0
pickleshare==0.7.5
pillow==10.2.0
pkginfo==1.8.2
pkgutil_resolve_name==1.3.10
platformdirs==2.5.2
pluggy==1.0.0
prometheus-client==0.19.0
prompt-toolkit==3.0.43
protobuf==3.20.1
psutil==5.9.8
pure-eval==0.2.2
py==1.11.0
pyansys-sphinx-theme==0.2.3
pyasn1==0.4.8
pyasn1-modules==0.2.8
pycparser==2.21
pydata-sphinx-theme==0.7.2
Pygments==2.11.2
pyparsing==3.0.7
pytest==7.1.2
pytest-cov==3.0.0
python-dateutil==2.8.2
python-json-logger==2.0.7
pytz==2022.1
pywin32==306
pywin32-ctypes==0.2.0
pywinpty==2.0.12
PyYAML==6.0.1
pyzmq==25.1.2
readme-renderer==34.0
referencing==0.33.0
requests==2.31.0
requests-toolbelt==0.9.1
rfc3339-validator==0.1.4
rfc3986==2.0.0
rfc3986-validator==0.1.1
rpds-py==0.17.1
rsa==4.8
Send2Trash==1.8.2
six==1.16.0
sniffio==1.3.0
snowballstemmer==2.2.0
soupsieve==2.3.2.post1
Sphinx==4.5.0
sphinx-copybutton==0.5.0
sphinxcontrib-applehelp==1.0.2
sphinxcontrib-devhelp==1.0.2
sphinxcontrib-htmlhelp==2.0.0
sphinxcontrib-jsmath==1.0.1
sphinxcontrib-qthelp==1.0.3
sphinxcontrib-serializinghtml==1.1.5
stack-data==0.6.3
terminado==0.18.0
tinycss2==1.2.1
toml==0.10.2
tomli==2.0.1
tomli_w==1.0.0
tornado==6.4
tox==3.25.0
tqdm==4.63.0
traitlets==5.14.1
twine==3.8.0
types-python-dateutil==2.8.19.20240106
typing_extensions==4.9.0
tzdata==2023.3
uri-template==1.3.0
uritemplate==4.1.1
urllib3==1.26.8
virtualenv==20.14.1
wcwidth==0.2.13
webcolors==1.13
webencodings==0.5.1
websocket-client==1.7.0
zipp==3.7.0
```

hiro727, 2024-03-20:
default max message size is 4MB. needs to be increased or IDs need to be sent back in chunks

hiro727, 2024-05-16:
proto support added https://github.com/ansys/ansys-api-edb/pull/20

closed 2025-05-12 by drewm102