ansys/pyedb-core #371: Performance issues on grpc api

svandenb-dev, 2024-03-01:
### 🔍 Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### 🐞 Description of the bug

I connected the unit test from pyedb. First test from passed but was slow. Since I can switch the test from .NET to grpc, using the same code, same project here is a picture from the difference directly printed from the code execution (time is in second)

same command from unit test grpc and .NET 
selected_pins = [pin for pin in self._pedb.components.components[ref].pins.values()
                             if pin.net_name in net_list and pin.is_pin]

PyEDB INFO: EDB initialized.
PASSED [100%]Fetching pins belonging to net DDR4_DQS0_P and DDR4_DQS0_N
time_elapsed grpc api : 62.692482233047485
Fetching pins belonging to net DDR4_DQS0_P and DDR4_DQS0_N
time_elapsed grpc api : 62.796724796295166

PyEDB INFO: EDB initialized.
PASSED [100%]Fetching pins usinh .NET api
time_elapsed .NET api : 0.07809185981750488
Fetching pins usinh .NET api
time_elapsed .NET api : 0.0821077823638916

this command is retrieving pins (e.g. padstack instances with checking the net_name)

### 📝 Steps to reproduce

NA

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

NA

### 🐍 Which Python version are you using?

3.9

### 📦 Installed packages

```shell
NA
```

hiro727, 2024-03-20:
the number of queries is the real bottleneck here. two options:
1. do filtering on backend
```py
pins = components.pins(net=net_list, is_pin: True)
```
2. send back all attributes of `PadstackInstance` to avoid n + 1 queries
```py
components.pins
#=> [
#   PadstackInstance({ id: 1, net_name: "net1", is_pin: True }),
#   PadstackInstance({ id: 2, net_name: "net2", is_pin: False })
#   # etc
# ]
```

svandenb-dev, 2024-03-21:
closing this issue. It seems to be related to pyedb code. So I have to investigate on my side before performing addionnal tests.

closed 2024-03-21 by svandenb-dev