ansys/pyedb-core #372: When RPC server is already running opening an EDB fails ->need to kill the process and restart

svandenb-dev, 2024-03-01:
### 🔍 Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### 🐞 Description of the bug

First init with starting RPC server works fine we can open EDB file. Then after closing I want to reopen another one, the init is detecting the server is running as I got the pid but then if I open the EDB it fails. As a work round I will kill and restart the sever but we can't open multiple EDB and if the server is running remotely killing the sever will impact all users connected.

### 📝 Steps to reproduce

NA

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

NA

### 🐍 Which Python version are you using?

3.9

### 📦 Installed packages

```shell
NA
```

hiro727, 2024-03-20:
running `change_capacitor_models.ipynb` multiple time works fine. (each run opens a db, performs operations, then closes it)

I tested the example as is (disconnect current session after each run) and without disconnecting from session (close db and reopen a new one right away) and didn't see issues

@svandenb-dev can you post exact repro step please?

svandenb-dev, 2024-03-20:
@hiro727 we should check together unless I missed something on how the server works. I can reproduce on my unit test

svandenb-dev, 2024-03-21:
connection to existing session fixed the issue