ansys/pyedb-core #373: PadstackInstanceTerminal.create() docstring wrong

svandenb-dev, 2024-03-06:
### Description of the modifications

pyedb code:

    def create(self, padstack_instance, name=None, layer=None, is_ref=False):
        """Create an edge terminal.

        Parameters
        ----------
        prim_id : int
            Primitive ID.
        point_on_edge : list
            Coordinate of the point to define the edge terminal.
            The point must be on the target edge but not on the two
            ends of the edge.
        terminal_name : str, optional
            Name of the terminal. The default is ``None``, in which case the
            default name is assigned.
        is_ref : bool, optional
            Whether it is a reference terminal. The default is ``False``.
        Returns
        -------
        Edb.Cell.Terminal.EdgeTerminal

online doc

[image]

hiro727, 2024-03-20:
@svandenb-dev your pyedb code shows `EdgeTerminal` whereas the doc shows `PadstackInstanceTerminal`.
doc for each method seems correct. can you clarify?

svandenb-dev, 2024-03-20:
@hiro727 I will check again maybe this was a mistake

drewm102, 2024-10-11:
@svandenb-dev It looks like the docstring is correct. I'm going to close this out for now

closed 2024-10-11 by drewm102