ansys/pyedb-core #376: Checking for port used before starting server session

svandenb-dev, 2024-03-21:
### 📝 Description of the feature

Checking port in case server is already running when start_session is called

hiro727, 2024-04-10:
need to add health check api and query against it to make sure the service on the specified port is indeed edb server

cc @drewm102

drewm102, 2024-04-18:
@chenchienjacklin As per our discussion today, we should modify the behavior of `launch_session(...)` to be the following:

- If there is no active session and no server executable listening on the provided port, start the server executable and attach to it. 
- If there is no active session and there is a server executable listening on the provided port, attach to it
- If a session already exists on the provided port, reuse it
- If a session already exists but it is on a different port than requested, throw an error

closed 2024-05-02 by chenchienjacklin