ansys/pyedb-core #384: PadstackInstance.component getter return broken component object

svandenb-dev, 2024-04-10:
### 🔍 Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### 🐞 Description of the bug

[image]

[image]

[image]

I checked the pin has net and all looks ok.

### 📝 Steps to reproduce

NA

### 💻 Which operating system are you using?

Windows

### 🐍 Which Python version are you using?

3.10

### 📦 Installed packages

```shell
NA
```

hiro727, 2024-04-10:
`ConnObj.component` method is incorrectly returning `ComponentGroup` instead of `Component`

drewm102, 2024-10-10:
@svandenb-dev Is the pin in question part of a component? If not, the behavior you are seeing is expected since the ComponentGroup returned by the component property is null and therefore throws an exception when a method is called on it

svandenb-dev, 2024-10-14:
We can close issue. This was open long time ago.

closed 2024-10-14 by svandenb-dev