ansys/pyedb-core #428: BUG: ComponentGroup.component return trace back

svandenb-dev, 2024-10-04:
### 🔍 Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### 🐞 Description of the bug

groups[0].component

[image]

### 📝 Steps to reproduce

from pyedb.generic.general_methods import generate_unique_folder_name
import pyedb.misc.downloads as downloads
import ansys.edb.core.database as database
from ansys.edb.core.session import launch_session
base_path = r"C:\Program Files\AnsysEM\v242\Win64"
port = 50051

temp_folder = generate_unique_folder_name()
targetfile = downloads.download_file("edb/ANSYS-HSD_V1.aedb", destination=temp_folder)

session = launch_session(base_path, port_num=port)

edb = database.Database.open(db_path=targetfile, read_only=False)
layout = edb.top_circuit_cells[0].layout
groups = edb.top_circuit_cells[0].layout.groups
groups[0].component

### 💻 Which operating system are you using?

Windows

### 🐍 Which Python version are you using?

3.9

### 📦 Installed packages

```shell
NA
```

drewm102, 2024-10-07:
@svandenb-dev I can't reproduce this bug. Are you sure you are using the most recent version of pyedb-core? And what AEDT install are you using? I was testing with R25.2 which should be pretty much the same as R25.1 at this point.

drewm102, 2024-10-14:
@svandenb-dev Ah, ok I think I see what the issue is. The `component` property returns the `ComponentGroup` that the object in question is part of. In this case, that object is a `ComponentGroup` which is not part of another `ComponentGroup`. As a result, the object returned by the `component` property is null (note that the `is_null` property returns true). An assertion is thrown when a null object is accessed which is why you are seeing all the tracebacks in PyCharm. I'm going to close this out for now but let me know if there us a different behavior that is expected.

closed 2024-10-14 by drewm102