ansys/pyedb-core #429: BUG: PadstackInstanceTerminal.padstack_instance property fails

svandenb-dev, 2024-10-07:
### 🔍 Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### 🐞 Description of the bug

When a Padstackinstance is defined and we use the delete() method. the PadstackInstance is also removed. Therefore I tried the cast the PadstackInstance from PadstackinstanceTerminal to instantiate a new Padstackinstance but it fails. 

I am not sure if this is a bug from PadstackInstanceTerminal.delete() Are we supposed to loose the PadstackInstance too ?

[image]

[image]

### 📝 Steps to reproduce

NA

### 💻 Which operating system are you using?

Windows

### 🐍 Which Python version are you using?

3.10

### 📦 Installed packages

```shell
Na
```

drewm102, 2024-10-07:
@svandenb-dev From my testing, deleting a PadstackInstanceTerminal does not delete the PadstackInstance it resides on. Can you provide the code you are using so I can better understand of the behavior you are seeing? Thanks!

svandenb-dev, 2024-10-09:
from @drewm102  You can ignore my comment, I was able to reproduce and am looking into a solution

PR #436 (2024-10-11): FIX: Miscellaneous fixes
closes #417, closes #431, closes #434, closes #429, closes #427, closes #426, closes #418, closes #414
closed 2024-10-11 by drewm102