ansys/pyedb-core #433: BUG: Cell.add_simulation_setup() missing

svandenb-dev, 2024-10-10:
### 🔍 Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### 🐞 Description of the bug

Cell.add_simulation_setup() is missing despite delete_simulation_setup is here.
Also Cell.simulation_setup doesn't have setter otherwise we could pass a list of Simulation setups.

This is required when we clip a Cell, simulation setups are not duplicated so we need to explicitly copy the setup from parent to the clipped cell.

### 📝 Steps to reproduce

NA

### 💻 Which operating system are you using?

Windows

### 🐍 Which Python version are you using?

3.10

### 📦 Installed packages

```shell
NA
```

drewm102, 2024-10-10:
@svandenb-dev The syntax for creating sim setups is different in pyedb-core. There is a `create` method on the simulation setup classes for creating them in a given cell. For example:

```
db = Database.create("test.aedb")
test_cell = Cell.create(db, CellType.CIRCUIT_CELL, "test_cell")
hfss_sim_setup = HfssSimulationSetup.create(test_cell, "test_sim_setup")
```

closed 2024-10-10 by drewm102

svandenb-dev, 2024-10-15:
ok thanks it works