ansys/pyedb-core #440: setup.settings.general.single_frequency_adaptive_solution.adaptive_frequency setter doesn't take value

svandenb-dev, 2024-10-12:
### 🔍 Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### 🐞 Description of the bug

[image]

setup.settings.general.single_frequency_adaptive_solution.adaptive_frequency = "1GΗz"
Property accept the value but using the getter does not show the updated value.

### 📝 Steps to reproduce

NA

### 💻 Which operating system are you using?

Windows

### 🐍 Which Python version are you using?

3.10

### 📦 Installed packages

```shell
NA
```

svandenb-dev, 2024-10-14:
I think this issue can be extended to all HfssSimulationSetup sub classes setters

drewm102, 2024-10-14:
@svandenb-dev For the changes made to the `single_frequency_adaptive_solution` to be stored in the sim setup object, you must set the `single_frequency_adaptive_solution` property with the `SingleFrequencyAdaptiveSolution` that has had the changes applied to it. For example, to change the `adaptive_frequency` property, you do this:

```
sfs = setup.settings.general.single_frequency_adaptive_solution
sfs.adaptive_frequency = "1GΗz"
setup.settings.general.single_frequency_adaptive_solution = sfs
```

closed 2024-10-14 by drewm102

svandenb-dev, 2024-10-15:
[like]  Simon Vandenbrouck reacted to your message:

[External Sender]