ansys/pyedb-core #441: HfssSimulationSetup.SweepData adding sweep

svandenb-dev, 2024-10-13:
### 🔍 Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### 🐞 Description of the bug

I created a simulation setup it looks ok.

[image]

then a SweepData

[image]

When I try using the setter to create the sweep I got this error

[image]

### 📝 Steps to reproduce

here is my code 

from ansys.edb.core.simulation_setup.hfss_simulation_setup import HfssSimulationSetup as GrpcHfssSimulationSetup
        from ansys.edb.core.simulation_setup.simulation_setup import SweepData as GrpcSweepData
        if not name:
            name = generate_unique_name("HFSS_pyedb")
        setup = GrpcHfssSimulationSetup.create(self._pedb.active_cell, name)
        start_freq = self._pedb.number_with_units(start_freq, "Hz")
        stop_freq = self._pedb.number_with_units(stop_freq, "Hz")
        if distribution.lower() == "linear":
            distribution = "LIN"
        elif distribution.lower() == "linear_count":
            distribution = "LINC"
        elif distribution.lower() == "exponential":
            distribution = "ESTP"
        elif distribution.lower() == "decade_count":
            distribution = "DEC"
        elif distribution.lower() == "octave_count":
            distribution = "OCT"
        else:
            distribution = "LIN"
        sweep_name = f"sweep_{len(setup.sweep_data) + 1}"
        sweep_data = [
            GrpcSweepData(
                name=sweep_name, distribution=distribution, start_f=start_freq, end_f=stop_freq, step=step_freq
            )
        ]
        if discrete_sweep:
            sweep_data[0].type = sweep_data[0].type.DISCRETE_SWEEP
        for sweep in setup.sweep_data:
            sweep_data.append(sweep)
        setup.sweep_data = sweep_data
        return HfssSimulationSetup(self._pedb, setup)

### 💻 Which operating system are you using?

Windows

### 🐍 Which Python version are you using?

3.10

### 📦 Installed packages

```shell
NA
```

drewm102, 2024-10-14:
@svandenb-dev It looks like the `step` property in the `SweepData` object is a `float`. It must be a `string`. I believe that is where the error is coming from since the code attempts to concatenate the start, stop, and step strings of the `SweepData` object together when creating the frequency string of the `SweepData` object. Can you try fixing this and letting me know if it resolves the issue?

svandenb-dev, 2024-10-15:
@drewm102 I confirm this was the issue thanks !! closing.

closed 2024-10-15 by svandenb-dev