ansys/pyedb-core #444: How to assign Mesh operation ?

svandenb-dev, 2024-10-14:
### 🔍 Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### 🐞 Description of the bug

My mesh operation is created 
mop = GrpcLengthMeshOperation(name=name,
                                      net_layer_info=net_layer_op,
                                      refine_inside=refine_inside,
                                      mesh_region=mesh_region,
                                      max_length=max_length,
                                      restrict_max_length=restrict_length,
                                      restrict_max_elements=restrict_elements,
                                      max_elements=max_elements)

[image]

[image]

[image]

Mesh operation remains empty

[image]

also tried this

[image]

### 📝 Steps to reproduce

NA

### 💻 Which operating system are you using?

Windows

### 🐍 Which Python version are you using?

3.10

### 📦 Installed packages

```shell
NA
```

drewm102, 2024-10-14:
@svandenb-dev You are correct that mesh ops should be set like this:

```
self.mesh_operations = [mop]
```

However, `max_elements` should be a `str` not an `int`. Changing it to an `str` will fix this issue.

closed 2024-10-14 by drewm102

svandenb-dev, 2024-10-15:
@drewm102 I confirm it is working now