ansys/pyedb-core #450: Bondwire.set_start_elevation()

svandenb-dev, 2024-10-16:
### 🔍 Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### 🐞 Description of the bug

set_start_elevation requires CellInstance and layer as argument. However CellInstance is optional, EDB .NET accept the current opened Cell as argument.

When I try to change the bondwire elevation I got this error.

[image]

[image]

[image]

### 📝 Steps to reproduce

NA

### 💻 Which operating system are you using?

Windows

### 🐍 Which Python version are you using?

3.11

### 📦 Installed packages

```shell
NA
```

drewm102, 2024-10-21:
@svandenb-dev I don't think that passing the active `Cell` object is an option for this API in the .NET API. In the .NET API, the context parameter must either be a `CellInstance` or `null`. In the edb rpc API, it must be either a `CellInstance` or `None`. Can you try passing `None`? I think that should resolve this issue.

drewm102, 2025-04-29:
@svandenb-dev I'm closing this out because of my previous comment. Feel free to reopen it in the future if needed though

closed 2025-04-29 by drewm102

svandenb-dev, 2025-04-29:
[like]  Simon Vandenbrouck reacted to your message:

[External Sender]