ansys/pyedb-core #468: ComponentGroup.add_member() failing

svandenb-dev, 2024-11-11:
### 🔍 Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### 🐞 Description of the bug

I am creating a component
new_cmp = GrpcComponentGroup.create(self._active_layout, component_name, compdef.name)
[image]

I got the padstackinstance
[image]

[image]

Did I missed something ?

### 📝 Steps to reproduce

NA

### 💻 Which operating system are you using?

Windows

### 🐍 Which Python version are you using?

3.11

### 📦 Installed packages

```shell
NA
```

chenchienjacklin, 2024-11-13:
@svandenb-dev here is how I add a member
```
psi = PadstackInstance.create(
    self._layout,
    net,
    pin_name,
    padstack_def,
    position[0],
    position[1],
    Value(0),
    layer,
    layer,
    None,
    None,
)
psi.is_layout_pin = True
ComponentPin.create(component_def, psi.name)
component.add_member(psi)
```
can you try adding line `ComponentPin.create(component_def, psi.name)` ?

drewm102, 2024-11-15:
@svandenb-dev I believe this is the same way component pins are created/added to a component in the .NET API? I'm open to the idea of maybe automatically handling the step of creating the `ComponentPin` object, but do you think it would be better for that to be handled in the pyedb API since that is a high level wrapper around the pyedb-core API? Thanks!

svandenb-dev, 2024-11-18:
@drewm102 I think this would need to be handled at the wrapping layer like you said. Adding the Components Pins is done at the definition level, so to my understanding ComponentPin should be create when we create the ComponentDefinition then we place pin "1", "2".... When we add the member the PadstackInstance will have to be renamed with the existing ComponentPin name. The code shown by @chenchienjacklin will assign the PadstackInstance name to all component sharing the same defintion, do you agree ?

closed 2025-05-13 by chenchienjacklin