ansys/pyedb-core #485: BUG: Path.get_product_property(ProductIdType.DESIGNER, 1) failing

svandenb-dev, 2025-02-06:
### 🔍 Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### 🐞 Description of the bug

We are using this method in PyEDB to retrieve the path name in AEDT, it works for other primitives but seems to be failing with Path.

### 📝 Steps to reproduce

NA

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

NA

### 🐍 Which Python version are you using?

3.11

### 📦 Installed packages

```shell
NA
```

chenchienjacklin, 2025-03-05:
Hi @svandenb-dev I'm not able to reproduce the issue, could you provide the script you use?
Here is the script I use to get all the names

```
from ansys.edb.core.database import Database, ProductIdType
from ansys.edb.core.layout.cell import Cell
from ansys.edb.core.session import session

ANSYS_EM_ROOT = r"C:\Program Files\AnsysEM\v252\Win64"
EDB_PATH = r"C:\Program Files\AnsysEM\v252\Win64\Examples\HFSS 3D Layout\Signal Integrity\SSN.aedb"
with session(ANSYS_EM_ROOT, 50051):
    try:
        db = Database.open(EDB_PATH, False)
        cell: Cell = db.circuit_cells[0]
        layout = cell.layout
        primitives = layout.primitives
        names = [prim.get_product_property(ProductIdType.DESIGNER, 1) for prim in primitives]
        print(names)
    except Exception as e:
        raise e
    finally:
        db.close()
```

And here is the result
```
['line_0', 'line_1', 'line_2', 'line_3', 'line_4', 'line_5', 'line_6', 'line_7', 'line_8', 'line_9', 'line_10', 'line_11', 'line_12', 'line_13', 'line_14', 'line_15', 'line_16', 'line_17', ...]
```

svandenb-dev, 2025-03-28:
This issue has been resolved.

closed 2025-03-28 by svandenb-dev